umit-iace/tool-pywisp #63: Canvas draw command is too slow

albflo, 2019-09-04:
In my example of the heavy chain, I got drawing times of 30 - 40 ms for my visualization. So I looked now of a better way to do it and to speed up my visualization. I found out that the canvas command draw_idle() handles the drawing way better and I got down to 5 - 10 ms, but weirdly it also calls the draw command. It simply checks if someone is drawing and if not, it calls the draw command. Could be that qt handles the draw_idle better than the draw command.

Someone could test that in another visualization and if the outcome is the same, you could change the drawing command in the examples to draw_idle.

albflo, 2019-09-04:
Tested the same approach with the current tcp example and it shows the exact same behavior. So it is faster to use draw_idle instead of draw.

Would be nice if @gruberth could validate this and add those changes to master. Please close this issue afterwards.

TanDro, 2019-09-25:
I have it tested at the different examples, and `draw_idle` is definitely as `draw`. I have update update the examples and the documentation to recomment `draw_idle`.

closed 2019-09-25 by TanDro