umit-iace/tool-pywisp #65: unpack buffer size error

simbach, 2019-10-22:
When I run an experiment and switch during running to another experiment with more data points. pywisp crashes with an error that the buffer size for unpack is too small

jhalmen, 2019-10-22:
this sounds like a user error.
Look at the function that unpacks these datapoints that are not in your original experiment. specifically at the `struct.unpack(...)` call.  the data buffer you give this function must be precisely the length that your format string requires. [the documentation](https://docs.python.org/2/library/struct.html) has more info.
I expect there to be an error in your format string.

simbach, 2019-10-23:
I don't think that its a user error. It also occurs sometimes when an experiment is stopped. Then all data is lost. It would be nice if you can look on this today. 

[image]

closed 2019-10-29 by TanDro