umit-iace/tool-pywisp #77: Arduino example bug

tappeluk, 2020-03-31:
In der klasse Transport.cpp:
In Zeile 54 this->bActivateExperiment); auf true gesetzt werden, damit die komuniktion mit python funktioniert?

In server.ino: 
In Zeile eine nicht definierte Funktion sei(); aufgerufen. 
In Zeile 13 ist ein typo unterlaufen unsigend.

jhalmen, 2020-03-31:
> this->bActivateExperiment); 

ich frag mich was da passiert ist :))

> nicht definierte Funktion sei();

`sei()` aktiviert in einem normalen avr die interrupts global wieder, nachdem einer aufgerufen wurde. was fuer einen mikrocontroller verwendest du?

danke fuer's reporten! 
nehmen gerne auch patches naechstes mal xD

tappeluk, 2020-03-31:
Danke! dann das nächste mal einen patch!
Aha ich verwende den due dieser ist ein arm daher kannte dieser vermutlich das sei() nicht. bin noch etwas beim experimentieren und würde die examples auch für das due erweitern?

jhalmen, 2020-03-31:
kurzes [googlen](https://github.com/MarkusLange/OneWireSlave/issues/3) zeigt:
anstatt von `sei()` einfach `interrupts()` verwenden, dann sollte es passen.

funktioniert der code auch ohne die zeile?

TanDro, 2020-03-31:
@jhalmen jetzt laufen alle Beispiele. @tappeluk ja bitte, wenn es funktioniert immer her damit.

jhalmen, 2020-03-31:
@TanDro wenn man `interrupts()` anstatt `sei()` verwendet, funktioniert's sowohl fuer avr als auch fuer arm mikrocontroller, ich glaube die unterordner braucht es nicht

TanDro, 2020-03-31:
ich vermute mal die Timer laufen auch anders, oder @tappeluk kann man fürn Arm die selbe Lib wie für nen Avr nutzen?

TanDro, 2020-03-31:
glaube https://github.com/ivanseidel/DueTimer wäre hier adäquat, hab leider keinen due im labor gefunden zum schnellen testen

closed 2020-05-06 by TanDro