umit-iace/tool-pywisp #82: Slider Konstruktor by usage with default.sreg

ditscheey, 2020-08-20:
Creating sliders in the gui works, decimal points and all parameters are usable.

Copying the created remote source of  the working **remote-widget** produces following error code:

[image]

passing an extra **startValue**-Parameter via *default.sreg* won't get passed.
**msg** still only contains 9 elements. 

quick fix: 
[image]

jhalmen, 2020-08-20:
for future reference: 
when reporting anything to do with text, please copy and paste the text itself instead of sharing screenshots where one has to look for the important text oneself.
github offers the possibility to paste this as 
```
preformatted 
    text
```
 patches are best posted in your own fork (from where the fix can be directly merged if it is acceptable), or in separate [gist](https://gist.github.com)s if it is just a proof of concept which you don't want to work on to bring it into a mergeable state.
posting a screenshot of a patch just puts the burden on the maintainer to redo the patch on his local machine before he is able to test it

jhalmen, 2020-09-02:
ok i just hit this as well.

the problem is, that the remote init code doesn't look at the defined modules and their `publicSettings`, it expects all these Settings to be redefined in the `default.sreg`.

so with my `testbench.py` defining:
``` python
class Leader(ExperimentModule):
    publicSettings = OrderedDict([("Manual", 0),
                                  ("Speed", 0),
                                  ("Rotation", 0)])
```
and my `default.sreg`:
```
- Name: Test

  Leader:

  Remote:
    Lead - Speed:
      Module: Leader
      Parameter: Speed
      maxSlider: '1'
      minSlider: '-1'
      shortcutMinus: Down
      shortcutPlus: Up
      stepSlider: '0.1'
      widgetType: Slider
```
this will crash, as `Leader` has `NoneType`, and the code is looking at `Leader['Speed']` to set the initial value.

if i just change the `default.sreg` to:
```
- Name: Test

  Leader:
    Manual: 1
    Speed: 0
    Rotation: 0

  Remote:
    Lead - Speed:
      Module: Leader
      Parameter: Speed
      maxSlider: '1'
      minSlider: '-1'
      shortcutMinus: Down
      shortcutPlus: Up
      stepSlider: '0.1'
      widgetType: Slider
```

this will finally work as intended.

@TanDro is this expected behaviour?
considering the defaults have to be defined in the class definitions anyway, and the `default.sreg` just offer the possibility of changing these defaults, i think it should be possible to use classes without having to redefine everything