umit-iace/tool-pywisp #89: default.sreg config names

jhalmen, 2020-09-09:
in #87 the issue about the configuration naming was raised, and silently ignored.
would it make sense to rename some of the configuration parameters?

TanDro, 2020-09-09:
which one next to timerTime?

jhalmen, 2020-09-10:
from https://github.com/umit-iace/tool-pywisp/pull/87#issuecomment-689366803
> * maybe rename the Time options to Timeout, as that's what they actually are
> * TimerTime -- says nothing about what it actually does. maybe Draw or Plot?

> @jhalmen in past we discuss this different time. Currently is used only to update the plot, right? Therfore, we can rename it to plotRefreshTime?

-> `PlotTimeout` & `HeartbeatTimeout` ?

TanDro, 2020-09-10:
Okay, we can do this.

TanDro, 2020-09-21:
see #96