umit-iace/tool-pywisp #104: Connection thread sleep problem

gruberth, 2021-06-10:
Habe jetzt rausgefunden warum bei mir Messungen über einer Minute am Pendelaufbau immer zu timeouts bzw. zu Abstürzen der SPS geführt haben.
Es liegt am
https://github.com/umit-iace/tool-pywisp/blob/2121583af9f6cf469af4c2a57210a574fb24cb27/pywisp/connection.py#L217
Aufruf. 
Die Genauigkeit und auch die minimal mögliche `sleep` time ist auf Linux und Windows komplett unterschiedlich. Wie auch [hier](https://stackoverflow.com/questions/1133857/how-accurate-is-pythons-time-sleep) auf stackoverflow besprochen wurde.

Ich habe folgenden Code von stackoverflow verwendet um dies zu bestätigen:

```
from datetime import datetime
import time

def check_sleep(amount):
    start = datetime.now()
    time.sleep(amount)
    end = datetime.now()
    delta = end-start
    return delta.seconds + delta.microseconds/1000000

duration = 0.001

sum = 0.0
for i in range(1000):
    sum += check_sleep(duration)

print("Expected sleep is %0.2fms" % (duration*1000))
print("Average sleep is %0.2fms" % sum)
```

Das Ergebnis auf Ubuntu war:

> Expected sleep is 1.00ms
Average sleep is 1.22ms

und auf Windows 10 mit WinPython:

> Expected sleep is 1.00ms
Average sleep is 15.69ms

jeweils mit Python 3.8.

Anscheinend geht zumindest bei mir nicht weniger als 15ms sleep auf Windows.

Habe heute das `time.sleep(0.001)` zum Testen auskommentiert. Das hat bzgl. der Messung einwandfrei funktioniert und die SPS ist auch nach vielen Minuten nicht abgestürzt, jedoch stockte dann manchmal kurz die GUI.

Habt ihr Ideen wie wir dieses Problem beheben oder umgehen könnten? (Außer auf Windows zu verzichten 😉)

TanDro, 2021-06-14:
Ich denke die einfachste Variante ist eine Fallunterscheidung einzubauen. Könntest du das folgende mal probieren:
```
if os.name == 'nt':
    time.sleep(0.05)
else:
    time.sleep(0.001)
```
Eine andere Alternative wäre es über die Win Api zu lösen
```
kernel32 = ctypes.windll.kernel32

# This sets the priority of the process to realtime--the same priority as the mouse pointer.
kernel32.SetThreadPriority(kernel32.GetCurrentThread(), 31)
# This creates a timer. This only needs to be done once.
timer = kernel32.CreateWaitableTimerA(ctypes.c_void_p(), True, ctypes.c_void_p())
# The kernel measures in 100 nanosecond intervals, so we must multiply .001 by 10000
delay = ctypes.c_longlong(.001 * 10000)
kernel32.SetWaitableTimer(timer, ctypes.byref(delay), 0, ctypes.c_void_p(), ctypes.c_void_p(), False)
kernel32.WaitForSingleObject(timer, 0xffffffff)
```

oder was ich noch gefunden habe:
```
def accurate_delay(delay):
    ''' Function to provide accurate time delay in millisecond
    '''
    _ = time.perf_counter() + delay/1000
    while time.perf_counter() < _:
        pass

delay = 0.001
t_start = time.perf_counter()
print('Wait for {:.0f} ms. Start: {:.5f}'.format(delay, t_start))

accurate_delay(delay)
```

gruberth, 2021-06-16:
> `time.sleep(0.05)`

das würde wahrscheinlich klappen, jedoch besteht dann immernoch das Problem, dass damit die `listen`-Schleife viel zu langsam ist.

Ich werde bei Gelegenheit mal die letzten Variante versuchen.

jhalmen, 2021-06-16:
Ich verstehe nicht wieso der längere Schlaf die SPS abstürzen lässt.

Wäre aber grundsätzlich dagegen den code mit mehr edge-cases zu erweitern.
Wir verwenden blocking reads auf dem socket, wenn also keine Daten ankommen wartet er automatisch bis die timeout Zeit um ist. brauchen solange das experiment laeuft also kein händisch eingebautes warten mehr.
und solang das experiment nicht läuft brauchen wir eh gar nicht lesen.

ich schlage vor:

- socket timeout erhöhen auf 10 ms
- in run methode nur schlafen wenn nötig

@gruberth versuch mal #105 

hab ich nicht getestet aber in meinem kopf sollte das funktionieren

gruberth, 2021-06-16:
Das abstürzen der SPS kommt wahrscheinlich vom overflow des `outbuffers`, wenn die gesendeten Pakete von pywisp nicht in angemessener Zeit ge-"ackt" werden.
Leider ist in B&R der Buffer nicht als Ringbuffer zu realisieren und die Größe des Buffers ist auch begrenzt. Sehe wenig Möglichkeiten um das Problem auf SPS Seite zu beheben.

> socket timeout erhöhen auf 10 ms

Ich habe das timeout in meiner Konfiguration auf 100ms.

Werde #105 demnächst testen.

gruberth, 2021-06-24:
Funktioniert mit #105 sehr gut.

commit 9292ee61 (2021-06-24): possibly fix #104 (#105)
closed 2021-06-24 by TanDro