umit-iace/tool-pywisp #133: Send bool followed by double

TanDro, 2023-03-29:
Sending the following sequenz leads to an hardware fault
PyWisp:
```
def getParams(self, data):
    payload = struct.pack('<?2d',
                          float(data[0]),
                          float(data[1]),
                          bool(data[2]),
                          )

    dataPoints = [
        {'id': 10,
         'msg': payload
         },
    ]

    return dataPoints
```
STM:
```
void setUserData(Frame &f) {
        auto manual = f.unpack<bool>();
        auto heaterPwm = f.unpack<double>();
        auto fanPwm = f.unpack<double>();
        hvac.bManual = manual;
        hvac.setHeaterPwm(heaterPwm);
        hvac.setFanPwm(fanPwm);
}
```
If it is changed to
PyWisp:
```
def getParams(self, data):
    payload = struct.pack('<2d?',
                          float(data[1]),
                          bool(data[2]),
                          float(data[0]),
                          )

    dataPoints = [
        {'id': 10,
         'msg': payload
         },
    ]

    return dataPoints
```
STM:
```
void setUserData(Frame &f) {
        auto heaterPwm = f.unpack<double>();
        auto fanPwm = f.unpack<double>();
        auto manual = f.unpack<bool>();
        hvac.bManual = manual;
        hvac.setHeaterPwm(heaterPwm);
        hvac.setFanPwm(fanPwm);
}
```
no problems.

jhalmen, 2023-03-29:
> ```
>     payload = struct.pack('<?2d',
>                           float(data[0]),
>                           float(data[1]),
>                           bool(data[2]),
>                           )
> ```
this in itself already has an issue:
the `?2d` means `bool, double, double`, but then the data you add is `double,double,bool`, so you'll never receive the data as you want it

> ```
>     payload = struct.pack('<2d?',
>                           float(data[1]),
>                           bool(data[2]),
>                           float(data[0]),
>                           )
> ```
here again `2d?` -> `double, double, bool`, but then the data you add is `double, bool, data`

you have to make sure to add the data in the right order, e.g.
for `data = [heater, fan, manual]`

```python

payload = struct.pack('<?2d',
    bool(data[2]),
    float(data[0]),
    float(data[1])
)
```
```c++
void setUserData(Frame &f) {
    f.unPack(hvac.bManual);
    hvac.setHeaterPwm(f.unpack<double>());
    hvac.setFanPwm(f.unpack<double>());
}
```

TanDro, 2023-04-03:
```
(gdb) backtrace -full
#0  0x080096fe in HardFault_Handler () at /home/jens/git/labor-hvac/Software/Mikrocontroller/stm/cmake-build-debug/_deps/tool-libs-src/stm/hal.cpp:26
No locals.
#1  <signal handler called>
No symbol table info available.
#2  Frame::unpack<double> (this=0x2007ff38) at /home/jens/git/labor-hvac/Software/Mikrocontroller/stm/cmake-build-debug/_deps/tool-libs-src/./comm/frameregistry.h:46
        ret = 1.7398508998480632e-310
#3  0x0800297a in Model::setUserData (this=0x2007ff98, f=...) at /home/jens/git/labor-hvac/Software/Mikrocontroller/stm/inc/model.h:66
        b = false
        heater = 2.2371152815169517e-154
        fan = 2.2372178840621952e-154
#4  0x08004242 in FrameRegistry::MethodHandler<Model>::handle (this=0x200023c0, f=...) at /home/jens/git/labor-hvac/Software/Mikrocontroller/stm/cmake-build-debug/_deps/tool-libs-src/./comm/frameregistry.h:80
No locals.
#5  0x08001410 in FrameRegistry::handle (this=0x200007a4 <support+104>, f=...) at /home/jens/git/labor-hvac/Software/Mikrocontroller/stm/cmake-build-debug/_deps/tool-libs-src/./comm/frameregistry.h:106
        c = 0x200023c0
#6  0x080022aa in Min::poll (this=0x2000073c <support>) at /home/jens/git/labor-hvac/Software/Mikrocontroller/stm/cmake-build-debug/_deps/tool-libs-src/./comm/min.h:261
No locals.
#7  0x080041f6 in Schedule::Recurring::Method<Min>::call (this=0x200022a0) at /home/jens/git/labor-hvac/Software/Mikrocontroller/stm/cmake-build-debug/_deps/tool-libs-src/./core/timed.h:61
No locals.
#8  0x080015c4 in Scheduler::run (this=0x200006f8 <k+12>) at /home/jens/git/labor-hvac/Software/Mikrocontroller/stm/cmake-build-debug/_deps/tool-libs-src/./core/schedule.h:61
        s = 0x200022a0
#9  0x080018a6 in Kernel::run (this=0x200006ec <k>) at /home/jens/git/labor-hvac/Software/Mikrocontroller/stm/cmake-build-debug/_deps/tool-libs-src/./core/kern.h:42
No locals.
#10 0x08002cc2 in main () at /home/jens/git/labor-hvac/Software/Mikrocontroller/stm/src/main.cpp:20
        m = {hvac = {input = {where = @0x2007ffd0, nested = 0x0, op = Later<Hvac::Input>::NOP}, state = {pipeTemp = 0, pipeTempDot = 0, heater = {pwm = 0}, fan = {pwm = 0}}, bManual = false}, trajType = {ref = {heater = 0, fan = 0}, traj = 0x20002300, type = TrajType::STEP}, fr = @0x200007a4, out = @0x20000788}
#11 0x080055d2 in Reset_Handler () at /home/jens/git/labor-hvac/Software/Mikrocontroller/stm/cmake-build-debug/_deps/stm32-cmsis-f7-src/Source/Templates/gcc/startup_stm32f767xx.s:100
No locals.
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

jhalmen, 2023-04-11:
I finally looked into this some more, and it's starting to look like `unalinged access` is causing this headache.
According to [this](https://developer.arm.com/documentation/ddi0337/h/ch03s04s02) unaligned accesses on the stm cpu are fine up to a `WORD` size, i.e. 4 bytes, but everything else will cause a fault. [Here](https://www.kernel.org/doc/Documentation/unaligned-memory-access.txt) i found more info on the concept of unaligned accesses. One quick solution would be to just use `float`s instead of `double`s (do we even need the higher precision?), but it doesn't solve the problem.
I'll rewrite the pack & unpack methods for the Frame without the possibility of unaligned accesses and hope that fixes your issue. I'll ping you again when it's ready

closed 2023-04-11 by TanDro